juliansteenbakker/nordic_dfu #1: Enhancement Request - DFU update using downloaded file

chris-otani, 2021-07-22:
Hi, Thank you for maintaining this package and migrating it to null safety. I have successfully integrated it into my project using a zip file in my flutter project just like the example project and it works as expected.

In my app I'd like to be able to get the zip firmware file from an API download rather than a local file so the user will always have access to the newest firmware. 

I can use an HTTP request to get the file contents but this package does not appear to have a method to update the firmware in this manner.

Would this be an enhancement that you can add or could you please point me in the right direction of some workaround?

Thank you!

juliansteenbakker, 2021-07-22:
Hello, it's funny that you ask because i've been wanting to implement the exact same thing for an app i'm developing. Just at the top of my head, isn't it possible to create an async function that will A: download the zipfile from the api and B: fire the `NordicDfu.startDfu` proces with the just aquired zip file path? I haven't really put any thought into it because i have this functionality planned in a sprint later the next month.

chris-otani, 2021-07-22:
That was my thoughts as well. I started to go down that path but it looks like it would take a few packages and extra permissions from the user to make it happen. I looked at the Nordic Android and IOS API and it looks like there is no method to pass a binary directly. It seems like something they could add easily. 

Let me know what you come up with if you implement something in your project.

rohitsangwan01, 2022-02-08:
@juliansteenbakker any progress update regarding this ??

juliansteenbakker, 2022-05-05:
What you have to do is use a tool like [flutter_downloader](https://pub.dev/packages/flutter_downloader) to download the zip to a public directory, like the downloads folder. You can give that path to this dfu utility in order to use the download.

closed 2022-05-05 by juliansteenbakker