juliansteenbakker/nordic_dfu #2: DFU Never Finishes

tpevahouse25, 2021-11-24:
I am having an issue with the nordic dfu service finishing. I have been using the flutter_nordic_dfu package for a while now with great success but since Android 12 has been released I have had to move to nordic_dfu. The dfu starts and in my on progress listener, I am printing out the percent. It will print out from 1% to 100% but then does nothing. My progress indicator shows that it has large jumps from 0 to 30 to 50 to 80 then 100 then stays at 100. The time it takes from 0% to 100% is only about 10 seconds when normally it would take about a minute and 30 seconds.  After a couple of minutes at 100% the app will crash. 

I am testing on a Google Pixel 6 with Android 12 and a Samsung Tab A with Android 11 and getting the same results. 
I can put this same code on an iPhone and it performs like it should and always has. 

Can you give me an idea why this might be happening and how i could fix it. 

Thank you.

juliansteenbakker, 2021-11-29:
Thanks for reporting. I will try and see if it works later this week.

tpevahouse25, 2021-12-07:
Have you found out anything on this yet?

juliansteenbakker, 2021-12-08:
No not yet, i do think this is an issue with the underlying nordic dfu android library. They are however actively improving this so i am waiting for a new version of it over at https://github.com/NordicSemiconductor/Android-DFU-Library

tpevahouse25, 2022-01-13:
Nordic has released an updated library and I was wondering if you have had the chance to work on this.

tpevahouse25, 2022-01-13:
I also found this post. Hope it helps. 
https://github.com/NordicSemiconductor/Android-DFU-Library/pull/313/files#diff-997602491f32cdd3bdde0f7b139809b82c9c0082cdd15b6ffa8b720628757717

tpevahouse25, 2022-01-14:
Thank you for working on this. I'll be looking forward for a package update on pub.dev.

juliansteenbakker, 2022-01-17:
@tpevahouse25 I have released v3.3.0. I hope it solves your problem.

tpevahouse25, 2022-01-17:
Unfortunately, that update didn't fix my issue. 
I am using a google pixel 6 with Android 12. My project is using Android sdk version 31. I have also made sure to add the new permissions to my manifest file. When I start the dfu service, it sits for about 3-5 seconds, then jumps to 47%. Then sits for another 3-5 seconds and jumps to 71%, 85%, 99%, then 100%. It will stay at 100% for about a minute then the dfu service will restart. 
Before upgrading my project to android sdk 31, the percentage would smoothly increase until finished successfully with no restarts. 
I have tried nordic's nrfConnect app to update and they have recently fixed this same issue with their app. It acted the same as described above until recently. Now it will successfully update a device without restarting the dfu service. 
Is there anything you can suggest me trying on my side?

tpevahouse25, 2022-01-17:
[image]

Here is the printout from when the DFU service finally failed after about 30 minutes of retrying.

Montvydas, 2022-04-08:
I got the same problem on my pixel 6. Although I noticed that recently even the nRF Connect app no longer works and outputs identical behaviour...

Montvydas, 2022-05-05:
Sadly it's still not working for me. But I think it might have something to do with the bootloader that I am using, it's [Adafruit bootloader](https://github.com/adafruit/Adafruit_nRF52_Bootloader), which I think is considered a legacy, since it's not a secure bootloader, thus maybe Nordic isn't serious about making the new SDK to work with it or simply they didn't test it...  
@juliansteenbakker can you share with us what kind of bootloader and which version are you using for testing? When trying to upload the firmware I get this kind of log and when it reaches 100 percent it stops:
```
I/DfuBaseService(  919): DFU service created. Version: 2.0.2
I/DfuBaseService(  919): Starting DFU service in foreground
I/DfuBaseService(  919): Connecting to the device...
D/BluetoothGatt(  919): connect() - device: F3:BB:99:46:63:A2, auto: false
D/BluetoothGatt(  919): registerApp()
D/BluetoothGatt(  919): registerApp() - UUID=9e6cedb3-b274-4323-ae62-95086f7f95aa
D/BluetoothGatt(  919): onClientRegistered() - status=0 clientIf=10
D/BluetoothGatt(  919): onClientConnectionState() - status=0 clientIf=10 device=F3:BB:99:46:63:A2
I/DfuBaseService(  919): Connected to GATT server
D/BluetoothGatt(  919): discoverServices() - device: F3:BB:99:46:63:A2
I/DfuBaseService(  919): Attempting to start service discovery... succeed
D/BluetoothGatt(  919): onSearchComplete() = Device=F3:BB:99:46:63:A2 Status=0
I/DfuBaseService(  919): Services discovered
I/DfuImpl (  919): Reading DFU version number...
I/DfuImpl (  919): Version number read: 0.8 -> Bootloader from SDK 9.0 or newer. Signature supported
W/DfuImpl (  919): Legacy DFU bootloader found
I/DfuImpl (  919): Requesting MTU = 517
D/BluetoothGatt(  919): configureMTU() - device: F3:BB:99:46:63:A2 mtu: 517
D/BluetoothGatt(  919): onConfigureMTU() - Device=F3:BB:99:46:63:A2 mtu=247 status=0
I/DfuImpl (  919): MTU changed to: 247
I/DfuImpl (  919): Enabling notifications...
D/BluetoothGatt(  919): setCharacteristicNotification() - uuid: 00001531-1212-efde-1523-785feabcd123 enable: true
I/DfuImpl (  919): Sending Start DFU command (Op Code = 1, Upload Mode = 4)
I/DfuImpl (  919): Sending image size array to DFU Packet (0b, 0b, 116240b)
I/DfuImpl (  919): Sending the Initialize DFU Parameters START (Op Code = 2, Value = 0)
I/DfuImpl (  919): Sending 14 bytes of init packet
I/DfuImpl (  919): Sending init packet (Value = 52-00-FF-FF-FF-FF-FF-FF-01-00-B6-00-A5-99)
I/DfuImpl (  919): Sending the Initialize DFU Parameters COMPLETE (Op Code = 2, Value = 1)
I/DfuImpl (  919): Sending Receive Firmware Image request (Op Code = 3)
I/flutter (  919): deviceAddress: F3:BB:99:46:63:A2, percent: 0
I/DfuImpl (  919): Uploading firmware...
I/flutter (  919): deviceAddress: F3:BB:99:46:63:A2, percent: 1
I/flutter (  919): deviceAddress: F3:BB:99:46:63:A2, percent: 2
I/flutter (  919): deviceAddress: F3:BB:99:46:63:A2, percent: 3
...
I/flutter (  919): deviceAddress: F3:BB:99:46:63:A2, percent: 100
```

juliansteenbakker, 2022-05-05:
@Montvydas I am using secure DFU which is working without problems. I think it would be better to open an issue on the android dfu github page https://github.com/NordicSemiconductor/Android-DFU-Library 
It does not seem like a problem that's caused in the flutter part of the plugin.

Satan-HuangZ, 2022-05-09:
Why ? fail to open file: No such file or directory

Satan-HuangZ, 2022-05-09:
> 

> Why ? fail to open file: No such file or directory

I solved it, thanks

tpevahouse25, 2022-05-10:
@juliansteenbakker Would you be able to tell me:
Library version you're using?
Which nRF SDK are you using?

juliansteenbakker, 2022-05-10:
> Which nRF SDK are you using?

https://github.com/juliansteenbakker/nordic_dfu/blob/master/android/build.gradle#L38

onurcoskun145, 2022-05-25:
Satan-HuangZ
how did you solve that problem? Can you help me?

onurcoskun145, 2022-05-25:
> > 
> 
> > Why ? fail to open file: No such file or directory
> 
> I solved it, thanks
how did did you solve it can you help me

onurcoskun145, 2022-05-25:
I/flutter (18078): PlatformException(File Error, File not found!, flutter_assets/assets/file.zip, null)

I am getting this error can anybody help ?

juliansteenbakker, 2022-05-25:
Did you add the file to your assets folder? And did you also add it in your pubspec.yaml?

juliansteenbakker, 2022-05-25:
`I/flutter (18078): PlatformException(File Error, File not found!, flutter_assets/assets/file.zip, null)`

Your path is not correct. You need to change `flutter_assets/assets/file.zip` to `assets/dfu_application.zip`

onurcoskun145, 2022-05-25:
the problem is that i dont have a folder named flutter_assets in my files

onurcoskun145, 2022-05-25:
If you mean i need to change path in main.dart file i have already done that

[image]
i really dont know where is flutter_assets/assets/file.zip path coming from

tpevahouse25, 2022-08-08:
I have been talking to Nordic about the issue with updating legacy bootloaders on android. I was told to use the default mtu of 23. I noticed that when using the nordic_dfu package is switches the mtu to 247. Is there a way to manually request an mtu using this package.
Also they have updated the android dfu library to version 2.1

juliansteenbakker, 2022-09-23:
I have released v5.1.0 of this package which updates both the android side aswell as the iOS side to the latest versions of the DFU library. The MTU is not something that can be controlled right now, so that would be a good addition imo.