juliansteenbakker/nordic_dfu #3: DefaultDfuProgressListenerAdapter callbacks will not called on Android 11

aBuder, 2021-12-08:
On my Android 11 (Google Pixel 3a)  `DefaultDfuProgressListenerAdapter` will not called.

```dart
      final dfuListener = DefaultDfuProgressListenerAdapter(
        onProgressChangedHandle: (
          deviceAddress,
          percent,
          speed,
          avgSpeed,
          currentPart,
          partsTotal,
        ) {
          dfuPercent = (percent ?? 0);
          log(
            'startOta: progressValue: $dfuPercent for device: $id',
            name: runtimeType.toString(),
            time: DateTime.now(),
          );
          addProgress(2 / 3 + (dfuPercent / 100) / 3);
        },
        onDfuProcessStartedHandle: (address) {
          log(
            "startOta: onDfuProcessStartedHandle for device: $id",
            name: runtimeType.toString(),
            time: DateTime.now(),
          );
        },
        onDeviceConnectedHandle: (address) {
          log(
            "startOta: onDeviceConnectedHandle for device: $id",
            name: runtimeType.toString(),
            time: DateTime.now(),
          );
        },
        onDfuCompletedHandle: (address) async {
          log(
            "startOta: onComplete for device: $id",
            name: runtimeType.toString(),
            time: DateTime.now(),
          );
          bluetoothUtil.isFirmwareUpdating = false;
          await SharedPrefUtils.setIsFirmware(false);
          downloadProgressStream.close();
        },
        onErrorHandle: (
          deviceAddress,
          error,
          errorType,
          message,
        ) {
          log(
            'startOta: onErrorHandle: $message for device: $id',
            name: runtimeType.toString(),
            time: DateTime.now(),
          );
          downloadProgressStream.addError(Exception(message));
        },
      );
      await NordicDfu.startDfu(
        id,
        path,
        progressListener: dfuListener,
      );
```

```log
Launching lib/main.dart on Pixel 3a in debug mode...
lib/main.dart:1
✓  Built build/app/outputs/flutter-apk/app-debug.apk.
W/FlutterActivityAndFragmentDelegate(11187): A splash screen was provided to Flutter, but this is deprecated. See flutter.dev/go/android-splash-migration for migration steps.
W/System  (11187): Ignoring header If-None-Match because its value was null.
Connecting to VM Service at ws://127.0.0.1:62530/6_vz42wy764=/ws
D/FRCPlugin(11187): Sending fetchTimeout: 60
D/FRCPlugin(11187): Sending fetchTimeout: 10
W/System  (11187): Ignoring header If-None-Match because its value was null.
D/FRCPlugin(11187): Sending fetchTimeout: 10
I/FLTFireBGExecutor(11187): Creating background FlutterEngine instance, with args: [--start-paused, --enable-dart-profiling]
I/FLTFireMsgService(11187): FlutterFirebaseMessagingBackgroundService started!
4
[log] [JOB]031a8e7a081e4c817319251ef8b42d00:triggered by opened_app
D/TransportRuntime.CctTransportBackend(11187): Making request to: https://firebaselogging-pa.googleapis.com/v1/firelog/legacy/batchlog
I/TransportRuntime.CctTransportBackend(11187): Status Code: 200
I/TransportRuntime.CctTransportBackend(11187): Content-Type: application/json; charset=UTF-8
I/TransportRuntime.CctTransportBackend(11187): Content-Encoding: gzip
I/flutter (11187): unhandled element filter; Picture key: AssetBundlePictureKey(bundle: PlatformAssetBundle#da405(), name: "packages/healy_ui/lib/assets/icons/Icons_Arrows_Right.svg", colorFilter: null)
I/AssistStructure(11187): Flattened final assist data: 644 bytes, containing 1 windows, 4 views
W/IInputConnectionWrapper(11187): beginBatchEdit on inactive InputConnection
W/IInputConnectionWrapper(11187): getTextBeforeCursor on inactive InputConnection
W/IInputConnectionWrapper(11187): getTextAfterCursor on inactive InputConnection
W/IInputConnectionWrapper(11187): getSelectedText on inactive InputConnection
W/IInputConnectionWrapper(11187): endBatchEdit on inactive InputConnection
W/IInputConnectionWrapper(11187): beginBatchEdit on inactive InputConnection
W/IInputConnectionWrapper(11187): endBatchEdit on inactive InputConnection
W/IInputConnectionWrapper(11187): beginBatchEdit on inactive InputConnection
W/IInputConnectionWrapper(11187): endBatchEdit on inactive InputConnection
4
[log] [JOB]031a8e7a081e4c817319251ef8b42d00:triggered by opened_app
2
W/System  (11187): Ignoring header X-Firebase-Locale because its value was null.
D/FirebaseAuth(11187): Notifying id token listeners about user ( a77355bb4fc1ec1c ).
D/FirebaseAuth(11187): Notifying auth state listeners about user ( a77355bb4fc1ec1c ).
W/DynamiteModule(11187): Local module descriptor class for com.google.android.gms.providerinstaller.dynamite not found.
I/DynamiteModule(11187): Considering local module com.google.android.gms.providerinstaller.dynamite:0 and remote module com.google.android.gms.providerinstaller.dynamite:0
W/ProviderInstaller(11187): Failed to load providerinstaller module: No acceptable module found. Local version is 0 and remote version is 0.
2
I/orld.healywatc(11187): The ClassLoaderContext is a special shared library.
W/ProviderInstaller(11187): Failed to report request stats: com.google.android.gms.common.security.ProviderInstallerImpl.reportRequestStats [class android.content.Context, long]
V/NativeCrypto(11187): Registering com/google/android/gms/org/conscrypt/NativeCrypto's 294 native methods...
W/orld.healywatc(11187): Accessing hidden method Ljava/security/spec/ECParameterSpec;->getCurveName()Ljava/lang/String; (greylist, reflection, allowed)

```

amar0k, 2022-03-10:
I have the same problem anyone have a fix for this?

ed-bonnet, 2022-03-10:
@amar0k 
Hello, I had this issue and my workaround was to remove flutter_downloader (used in my app to get the firmware file). 
The isolated process seems to be the issue. When flutter_downloader was called anywhere in the app, the DFU could not trigger any events.

amar0k, 2022-03-10:
@ed-bonnet Thanks for the reply, I am not using flutter_downloader i use S3 to get the firmware file. How did you manage to find what was preventing the call being made?

FluffyBunniesTasteTheBest, 2022-04-21:
Same issue on Android 10 after updating to version 4.0.0. No issue with version 3.3.0 though.

l984-451, 2022-04-29:
Did you find a fix for this? I have the same issue on Android and iOS.

NanaDada, 2022-04-29:
@l984-451 reverting back to version 3.3.0 was the only thing that worked. But even then, there was a problem with a Lenovo tablet on which DFU stubbornly refused to work because the DFU target wasn't found by the library. So, to fix this follow up problem on the Lenovo, you'll have to start a DFU scan and keep it running/scanning before starting the firmware update. This way it's going to work reliably.

closed 2022-05-05 by juliansteenbakker

GaelleJoubert, 2022-07-26:
I still have the same issue. No call backs is ever called, so no way for me to actually know when/if the dfu transfert has ended properly.

GaelleJoubert, 2022-08-25:
@juliansteenbakker ?

PR #36 (2022-05-05): imp: updated event channels and callbacks
Fixes #3