juliansteenbakker/nordic_dfu #19: Dfu device find by Name instead of ID

rohitsangwan01, 2022-02-24:
Hey can we do something like Start Dfu for a device with a custom Name , actually what happens is My Device id changes when it enters to Bootloader mode , so only thing i can do is change its name when enter to bootloader ,
and scan for all available device , get the device with that name , grab its id , and pass that id to this package

But can we do something like this with this package , like we can pass another parameter of custom name and deviceId ( OrignalId before Entering to Bootloader), later things can be handled by this package

basically in Android i can achive this by another Approach , 
for Example my device advertising in android
with id D5:59:39:3B:B3:59
and when its in Bootloader. its advertising with ID : D5:59:39:3B:B3:5A 

so basically adding 1 to the orignal Hexadecimal d559393bb359  + 1 => d559393bb35a
which is working fine , but for ios , no idea how to achieve this , 
like in case of IOS ,
Device ID : 6D8D8DC9-BA02-430F-4DC7-572E057D801B
DFU ID : 8100EEB2-F871-239E-80C2-03C03F613E58
Any Idead how can we detect in this case , or by adding 1 to DeviceID somehow can get us DfuID ??

juliansteenbakker, 2022-05-06:
This is quite specific for the IOS part and i think you can ask it better in the repository for the IOS DFU library: https://github.com/NordicSemiconductor/IOS-DFU-Library

What i do is that i scan for DFU packages while the device is in the bootloader and not before because the ID would not be the same.

closed 2022-05-06 by juliansteenbakker

lorenzosfarra, 2024-11-30:
Hi @rohitsangwan01,

I know this is an old issue, but I believe I’m encountering the same problem and could use some guidance.

How did you manage to resolve this cleanly?

Currently, the only method I’ve found is to call startDfu on the ID of the currently connected device. When the device reboots and enters bootloader mode, I scan for all available devices, identify the one with the desired name, retrieve its ID, and then pass that ID to this package.

It works, but it feels a bit cumbersome—do you have a more streamlined approach?

Thanks in advance!

rohitsangwan01, 2024-12-01:
Hey @lorenzosfarra i used similar approach