juliansteenbakker/nordic_dfu #23: sending information from kotlin to flutter

ThaddeusLim99, 2022-03-17:
I am currently using your plugin locally and trying to integrate it into an app that I am currently developing. The DFU update is working and I can read print statements that I inserted into the kotlin code in the debug console. However, unlike when I run the example app, the progressListener messages are not observable in the debug console and print statements that I put in the setMethodCallHandler are not being executed. I put print statements around invokeMethod in NordicDfuPlugin.kt  and the debug console detects it. I am not sure if this is a thread issue. I am currently testing using an Android 12 phone and VSCode.

Is it possible to help me with this? Thank you very much!

ThaddeusLim99, 2022-03-24:
Update: I got it to work (only Android so far) by using another repository that uses streams and eventChannels to communicate with the native code. (https://github.com/metaflowltd/flutter_ble_dfu). Still not sure of the exact reason for my issue but I will close this issue.

closed 2022-03-24 by ThaddeusLim99

GaelleJoubert, 2022-07-28:
> Update: I got it to work (only Android so far) by using another repository that uses streams and eventChannels to communicate with the native code. (https://github.com/metaflowltd/flutter_ble_dfu). Still not sure of the exact reason for my issue but I will close this issue.

I have the same issue than you, but the repository that you used is not available anymore ... Do you have another solution, Or do you know where I could find this repo ?