juliansteenbakker/nordic_dfu #45: DFU Transfert is working but No callback is never called.

GaelleJoubert, 2022-07-26:
I'm using the DFU, and the transfert works fine.
The only problem is : the call back are NEVER called, which means my app has NO WAY of nowing if/when the transfert is done.
With the prints, I see that it's going fine, but I'm wondering how Can I get these infos on my app ...

I use a Samsung A3 with Android 8, but I tried as well with a Pixel 4A 5G with Android 12 and I have the same issue.
I use Flutter 3.

Here is how I start the DFU :

```
await NordicDfu().startDfu(deviceID, file, fileInAsset: true,
    onDfuCompleted: (e){print("[DFU] Completed, $e");},
    onDeviceDisconnected: (e){print("[DFU] DIsconnected, $e");},
    onError: (e,a,b,c){print("[DFU] ERROR $e, $a, $b, $c");},
    onDfuAborted: (e){print("[DFU] Aborted $e");},
    onProgressChanged:(
        deviceAddress, percent, speed, avgSpeed, currentPart, partsTotal) {
          progressListener.progress(percent);
          print('[DFU] $percent%');
    });
```

All my prints I never done.
I still know it is working thanks to the console output :

```
I/DfuImpl ( 9193): Restarting to bootloader mode
I/DfuBaseService( 9193): Starting DFU service in foreground
D/Notification( 9193): allPendingIntents
D/BluetoothGatt( 9193): onClientConnectionState() - status=19 clientIf=8 device=EC:DB:86:00:02:FD
I/DfuBaseService( 9193): Action received: android.bluetooth.device.action.ACL_DISCONNECTED
D/BluetoothManager( 9193): getConnectionState()
D/BluetoothManager( 9193): getConnectedDevices
D/BluetoothGatt( 9193): close()
D/BluetoothGatt( 9193): unregisterApp() - mClientIf=8
[log] [2022-07-26 16:08:05.186651] [UPDATE] BLE Connection State : DeviceConnectionState.disconnected 
I/flutter ( 9193): EC.DB.86.FF.FE.00.02.FD. : NOT DETECTED - TIMEOUT : 29
D/Notification( 9193): allPendingIntents
I/DfuBaseService( 9193): Connecting to the device...
D/BluetoothGatt( 9193): connect() - device: EC:DB:86:00:02:FD, auto: false
D/BluetoothAdapter( 9193): isSecureModeEnabled
D/BluetoothGatt( 9193): registerApp()
D/BluetoothGatt( 9193): registerApp() - UUID=4519263b-5371-4b25-b983-e472d7812c35
D/BluetoothGatt( 9193): onClientRegistered() - status=0 clientIf=8
I/DfuBaseService( 9193): Action received: android.bluetooth.device.action.ACL_CONNECTED
D/BluetoothGatt( 9193): onClientConnectionState() - status=0 clientIf=8 device=EC:DB:86:00:02:FD
I/DfuBaseService( 9193): Connected to GATT server
I/DfuBaseService( 9193): Waiting 1600 ms for a possible Service Changed indication...
D/BluetoothGatt( 9193): discoverServices() - device: EC:DB:86:00:02:FD
I/DfuBaseService( 9193): Attempting to start service discovery... succeed
D/BluetoothGatt( 9193): onConnectionUpdated() - Device=EC:DB:86:00:02:FD interval=12 latency=0 timeout=600 status=0
D/BluetoothGatt( 9193): onConnectionUpdated() - Device=EC:DB:86:00:02:FD interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt( 9193): onConnectionUpdated() - Device=EC:DB:86:00:02:FD interval=12 latency=0 timeout=600 status=0
D/BluetoothGatt( 9193): onSearchComplete() = Device=EC:DB:86:00:02:FD Status=0
I/DfuBaseService( 9193): Services discovered
I/DfuImpl ( 9193): Reading Service Changed CCCD value...
I/DfuImpl ( 9193): Service Changed indications enabled
W/DfuImpl ( 9193): Secure DFU bootloader found
D/Notification( 9193): allPendingIntents
I/DfuImpl ( 9193): Requesting MTU = 517
D/BluetoothGatt( 9193): configureMTU() - device: EC:DB:86:00:02:FD mtu: 517
D/BluetoothGatt( 9193): onConfigureMTU() - Device=EC:DB:86:00:02:FD mtu=247 status=0
I/DfuImpl ( 9193): MTU changed to: 247
I/DfuImpl ( 9193): Enabling notifications...
D/BluetoothGatt( 9193): setCharacteristicNotification() - uuid: 8ec90001-f315-4f60-9fb8-838830daea50 enable: true
I/DfuImpl ( 9193): Setting object to Command (Op Code = 6, Type = 1)
I/flutter ( 9193): EC.DB.86.FF.FE.00.02.FD. : NOT DETECTED - TIMEOUT : 34
I/DfuImpl ( 9193): Command object info received (Max size = 512, Offset = 0, CRC = 00000000)
I/DfuImpl ( 9193): Sending the number of packets before notifications (Op Code = 2, Value = 0)
I/DfuImpl ( 9193): Creating Init packet object (Op Code = 1, Type = 1, Size = 141)
[... 68 lines cut ...]
D/Notification( 9193): allPendingIntents
I/DfuImpl ( 9193): Sending Calculate Checksum command (Op Code = 3)
I/DfuImpl ( 9193): Checksum received (Offset = 40960, CRC = CB0B8CB1)
I/DfuImpl ( 9193): Executing data object (Op Code = 4)
I/DfuImpl ( 9193): Creating Data object (Op Code = 1, Type = 2, Size = 3616) (11/11)
I/DfuImpl ( 9193): Uploading firmware...
D/Notification( 9193): allPendingIntents
I/DfuImpl ( 9193): Sending Calculate Checksum command (Op Code = 3)
I/DfuImpl ( 9193): Checksum received (Offset = 44576, CRC = 77EA64B6)
I/DfuImpl ( 9193): Executing data object (Op Code = 4)
I/flutter ( 9193): EC.DB.86.FF.FE.00.02.FD. : NOT DETECTED - TIMEOUT : 44
I/DfuImpl ( 9193): Transfer of 44576 bytes has taken 9637 ms
D/Notification( 9193): allPendingIntents
D/BluetoothGatt( 9193): onClientConnectionState() - status=19 clientIf=8 device=EC:DB:86:00:02:FD
W/DfuBaseService( 9193): Target device disconnected with status: 19
I/DfuBaseService( 9193): Cleaning up...
D/BluetoothGatt( 9193): cancelOpen() - device: EC:DB:86:00:02:FD
I/DfuBaseService( 9193): Action received: android.bluetooth.device.action.ACL_DISCONNECTED
D/BluetoothGatt( 9193): close()
D/BluetoothGatt( 9193): unregisterApp() - mClientIf=8
D/Notification( 9193): allPendingIntents
I/DfuBaseService( 9193): DFU service destroyed
```
Does any one has any idea to get the call back to work, or even just a workout to get info on how the transfert is going on ? I thought of rerouting the console output, but no idea on how to do that...
Thanks a lot !

GaelleJoubert, 2022-08-25:
Here are some updates about the issue :
I have found two "background process", which, if removed, make the callbacks things work again.
1- Firebase
At the very begining of our main we had this line :

`await Firebase.initializeApp(); `

We realize that if we commented it (with the 2nd background process also removed), the callbacks of the DFU would work again.
In the end, we put that line in the build method of the MyApp Widget instead of the main() and it works !

2 - Foreground  Task
In our app we have a foreground task running of a Bluetooth Scan using the Package flutter_foreground_task.
Of course before starting the DFU we stop that task. 
But if it has been launched even just one time, even after being stopped, it will cause the DFU callbacks to not being called.

If we comment the line that start this foreground task, meaning that the task is never started, then only the DFU callbacks are working.

We've tried many things, but so far we have not succeed to make the DFU callback working if the Foreground task has been started (and stoped). 

Our problem is that we actuelly need this two things in our app ...

I'm guessing something in the fact that another Activity or Isolate was running, is blocking the calls to the callbacks ... (the rest of the DFU is working, just the call backs are never called !!).

So if someone could find a fix for that with these extra infos, that would be amazing !
Thanks a lot.

GaelleJoubert, 2022-08-25:
Some more infos :
 I've try to use Work Mannager (https://pub.dev/packages/workmanager/) instead of flutter_foreground_task, to do the task I needed, and the result is the same. 
 If I don't launche the Task, the DFU calback are working well, And if I do launch it, the callbacks are never called :/
 For the sakes of the test, the task launched is completely empty, exept for a "print".
 
 Also I've noticed the option "startAsForegroundService" in the AndroidSpecificParameter, I've tried to set it to false, but the behavior doesn't change (meaning it is working only if no other background task has been launched).
 
 So it really seems like the library have an issue with other foreground tasks.

GaelleJoubert, 2022-08-31:
After a week of investigating the issue, and diving into the kotlin code, here is some more conclusions :

**About IOS**
I've tried my app on a Iphone, and it works well. 
My foreground service is running, and when I start the DFU, everything is working as expected, the callbacks are called properly.
So the problem seems to happen only on Android.

**About Android**
To better understand the lifecycle of a plugin and the kotlin part (I am rather new to that), I added some prints in the kotlin code. 
I have also played with the "workmannager" to better understand how the plugins with a "Foreground Task" works with each other. 

With the prints I noticed the Kotlin callbacks are correctly called.
BUT it seems it is the following line (calling the sink) who is not executed : 

`sink?.success(mapOf("onDeviceConnected" to deviceAddress))`

The `methodChannel` and `eventChannel` appears to be null.

So I decided to take a closer look at the functions  "onAttached" and "on detached".

Here is what I did, one step after another, to understand how those functions works, and at the same time, to make sure other plugins withe foreground task would works well (or not) with each other: 

1- Start my Foreground service with the plugin 1 (Flutter Foreground Task)
2- Stop the task
3- Start a foreground service with the plugin 2 (Work Mannager)
4- Stop that task
5- Start the DFU transfert

And here is what I noticed :
First, step 1 to 4 works well, as expected.
Step 5, as usual, the dfu is "working" but the callbacks are not called.

1- Trigger a call to ALL onAttached functions (meaning the ones from the DFU plugin, the Foreground_Task plugging and the Work mannager plugin).
2- Trigger a call to ALL onDettached functions
3- Trigger a call to ALL onAttached functions (meaning the ones from the DFU plugin, the Foreground_Task plugging and the Work mannager plugin).
4- Trigger a call to ALL onDettached functions
5- Trigger NO calls to either attached or detached functions.

**Note :** I I do only step 5 and NOT the other steps before, the DFU is working well and the callbacks are working.

CONCLUSION :
So for me the issue seemed to be the fact that the foreground tasks trigger a call to the onDetached function, but the onAttached function is never called again before the start of the DFU. The the methodChannel and eventChannel are null, and that's why the dart callbacks are not called anymore.

I tried to find a way to "force" onAttached to be called at the beggining of the plugging, but I am very new to this and I am not sure I understand properly how all of this works, ... so I did not succeed. I guess maybe it has something to do with the pluging needed to be an activity aware plugin ? I don't know. 

I also noticed a weird thing.
The onListen functiun is called at the launch of the DFU, and in that function methodChannel and eventChannel are not null. But right after that, for exemple in the onDeviceConnected, the methodChannel and eventChannel are null... So I guess something happened there, but I'm not quite sure what yet ...

But I hope these research can help someone to resolve this issue (@juliansteenbakker ?) and maybe spare them some time.

GaelleJoubert, 2022-09-05:
I solved the Issue !! 
@tr-asimpson if you want to fix it in your own repo while @juliansteenbakker  fix it in the library.

How to fix it :
 1- Open the file of the library `NordicDfuPlugin.kt` 

 Note : To access the file you can either : 
  a- find it with the explorer at the path :  `flutter\.pub-cache\hosted\pub.dartlang.org\nordic_dfu-5.0.1\android\src\main\kotlin\dev\steenbakker\nordicdfu\NordicDfuPlugin.kt`
  b- Open in in Android studio : Open in the project explorer "Flutter Plugins" -> nordic_dufu-5.0.1 -> android -> src.main -> kotlin.dev.steenbakker.nordicdfu -> NordicDfuPlugin
  
  2- In the function `onAttachedToEngine`, remove the line : `DfuServiceListenerHelper.registerProgressListener(mContext, mDfuProgressListener)` (line 40)
  
3- Copy that line in the` onListen()` function.

BOOM ! it fixes the issue, now the callbacks are called when we use the library with other plugin working with foreground services.

Here is the full code of the two modified functions : 

```
override fun onAttachedToEngine(binding: FlutterPluginBinding) {
        mContext = binding.applicationContext

        methodChannel = MethodChannel(binding.binaryMessenger, "dev.steenbakker.nordic_dfu/method")
        methodChannel!!.setMethodCallHandler(this)

        eventChannel = EventChannel(binding.binaryMessenger, "dev.steenbakker.nordic_dfu/event")
        eventChannel!!.setStreamHandler(this)

// THE FOLLOWING COMMENTED LINE NEED TO BE REMOVE
   //     DfuServiceListenerHelper.registerProgressListener(binding.applicationContext, mDfuProgressListener)
    }

 override fun onListen(arguments: Any?, events: EventChannel.EventSink?) {
//THE FOLLOWING LINE IS ADDED
        DfuServiceListenerHelper.registerProgressListener(mContext!!, mDfuProgressListener) // NEEDED !!
        this.sink = events
    }
```

tr-asimpson, 2022-09-06:
@GaelleJoubert I've been lurking and following. Your a legend.

Thanks buddy!

GaelleJoubert, 2022-09-06:
@tr-asimpson Thanks, I had no choices but to make it work haha !
You'll let me know if that fix works for you as well.

tr-asimpson, 2022-09-06:
@GaelleJoubert It works for me :) good job.

juliansteenbakker, 2022-09-22:
Sorry for the late response, but thanks for finding a fix! I've merged it and will be releasing it asap.

closed 2022-09-22 by juliansteenbakker