juliansteenbakker/nordic_dfu #57: Unable to track progress

nirjanmunshi, 2022-09-23:
This code is working fine on the device. Also, the OTA was successful. But I can not catch the progress coming in the `console`. Please let me now how to catch that. Thanks in advance.

Implementation

```dart
 var res =  await NordicDfu()
          .startDfu(
            device.id.id,
            otaController.otaFilePath,
            fileInAsset: false,
            onProgressChanged: (deviceAddress, percent, speed, avgSpeed, currentPart, partsTotal) {
              print(
                  'deviceAddress : $deviceAddress, percent: $percent, speed: $speed, avgSpeed: $avgSpeed, currentPart: $currentPart, partsTotal: $partsTotal');
            },
            onDeviceConnected: (address) {
              print('onDeviceConnected : $address');
            },
            onDeviceConnecting: (address) {
              print('onDeviceConnecting : $address');
            },
            onDeviceDisconnected: (address) {
              print('onDeviceDisconnected : $address');
            },
            onDfuAborted: (address) {
              print('onDfuAborted : $address');
            },
            onError: (address, error, errorType, message) {
              print('Error Occures: address: $address, error: $error, errorType: $errorType, message:$message');
            },
            onDfuCompleted: (address) {
              print('onDfuComplete: $address');
            },
            onFirmwareValidating: (address) {
              print('onFirmwareValidating: $address');
            },
            onEnablingDfuMode: (address) {
              print('onEnablingDfuMode: $address');
            },
            onDfuProcessStarted: (address) {
              print('onDfuProcessStarted: $address');
            },
            onDeviceDisconnecting: (address) {
              print('onDeviceDisconnecting: $address');
            },
            onDfuProcessStarting: (address) {
              print('onDfuProcessStarting: $address');
            },
          );
```

In `console` I am getting this

```shell
I/flutter (28926): DFU FOUND OTA FILE PATH :: /storage/emulated/0/Android/data/com.example.app/files/OTA/app_dfu_package_3.2.4.zip
I/DfuBaseService(28926): DFU service created. Version: 2.0.3
I/BluetoothAdapter(28926): BluetoothAdapter() : com.example.app
I/DfuBaseService(28926): Starting DFU service in foreground
D/CompatibilityChangeReporter(28926): Compat change id reported: 160794467; UID 11156; state: ENABLED
I/DfuBaseService(28926): Connecting to the device...
I/BluetoothAdapter(28926): STATE_ON
D/BluetoothGatt(28926): connect() - device: CA:89:A0:BD:AB:5B, auto: false
I/BluetoothAdapter(28926): isSecureModeEnabled
D/BluetoothGatt(28926): registerApp()
D/BluetoothGatt(28926): registerApp() - UUID=e949a72f-c290-4e87-93e0-9a40bb11a1ad
D/BluetoothGatt(28926): onClientRegistered() - status=0 clientIf=9
D/BluetoothGatt(28926): onClientConnectionState() - status=0 clientIf=9 device=CA:89:A0:BD:AB:5B
I/DfuBaseService(28926): Connected to GATT server
D/BluetoothGatt(28926): discoverServices() - device: CA:89:A0:BD:AB:5B
I/DfuBaseService(28926): Attempting to start service discovery... succeed
I/DfuBaseService(28926): Action received: android.bluetooth.device.action.ACL_CONNECTED
[... 377 lines cut ...]
I/DfuImpl (28926): Executing data object (Op Code = 4)
I/DfuImpl (28926): Creating Data object (Op Code = 1, Type = 2, Size = 4096) (72/73)
I/DfuImpl (28926): Uploading firmware...
I/DfuImpl (28926): Sending Calculate Checksum command (Op Code = 3)
I/DfuImpl (28926): Checksum received (Offset = 294912, CRC = 1BCFCDA5)
I/DfuImpl (28926): Executing data object (Op Code = 4)
I/DfuImpl (28926): Creating Data object (Op Code = 1, Type = 2, Size = 3320) (73/73)
I/DfuImpl (28926): Uploading firmware...
I/DfuImpl (28926): Sending Calculate Checksum command (Op Code = 3)
I/DfuImpl (28926): Checksum received (Offset = 298232, CRC = 988C64C1)
I/DfuImpl (28926): Executing data object (Op Code = 4)
I/DfuImpl (28926): Transfer of 298232 bytes has taken 66776 ms
D/BluetoothGatt(28926): onClientConnectionState() - status=0 clientIf=9 device=CA:89:A0:BD:AB:5B
I/DfuBaseService(28926): Disconnected from GATT server
W/(28926): Accessing hidden method Landroid/bluetooth/BluetoothGatt;->refresh()Z (unsupported, reflection, allowed)
D/BluetoothGatt(28926): refresh() - device: CA:89:A0:BD:AB:5B
I/DfuBaseService(28926): Refreshing result: true
I/DfuBaseService(28926): Cleaning up...
D/BluetoothGatt(28926): cancelOpen() - device: CA:89:A0:BD:AB:5B
D/BluetoothGatt(28926): close()
D/BluetoothGatt(28926): unregisterApp() - mClientIf=9
I/DfuBaseService(28926): Action received: android.bluetooth.device.action.ACL_DISCONNECTED
I/DfuBaseService(28926): DFU service destroyed

```

GaelleJoubert, 2022-09-27:
If no callback you give is ever called, it looks like the issue I raised some months agao, it is now fixed in the last version (5.1.0).
Check that you have that last version installed in your pubspeck.yaml :)

juliansteenbakker, 2022-09-28:
I have released v5.1.1. Can you please check if the issue still persists, otherwise feel free to comment.

closed 2022-09-28 by juliansteenbakker

matan-h, 2025-01-26:
I have the exact same issue with version 7.0.0. The `onProgressChanged` method is not being called, even though I see the `DfuImpl` logs and the firmware updates successfully. Reverting to version 6.2.0 resolves the issue.