juliansteenbakker/nordic_dfu #58: Last version won't build for IOS

GaelleJoubert, 2022-09-26:
I tried to update the package to 5.1.0.
No issue on Android, but IOS won't build.
I get the following print :

```
Could not build the precompiled application for the device.

Swift Compiler Error (Xcode): Initializer for conditional binding must have

Optional type, not 'DFUFirmware'

/Users/gaellejoubert/Documents/flutter/.pub-cache/hosted/pub.dartlang.org/nordic

_dfu-5.1.0/ios/Classes/SwiftNordicDfuPlugin.swift:105:14

Swift Compiler Error (Xcode): Call can throw, but it is not marked with 'try'

and the error is not handled

/Users/gaellejoubert/Documents/flutter/.pub-cache/hosted/pub.dartlang.org/nordic

_dfu-5.1.0/ios/Classes/SwiftNordicDfuPlugin.swift:105:29
```

I tried `Flutter clean`, I also tried to edit a bit the source code, but I know nothing about swift ...
Anybody has the same issue ?

GaelleJoubert, 2022-09-27:
Note : If I revert to the 5.0.1 version, I don't have this issue.

BenelliFurtado, 2022-09-27:
First, I had some other build issue

```
    [!] CocoaPods could not find compatible versions for pod "Flutter":
      In Podfile:
        Flutter (from `Flutter`)

        url_launcher_ios (from `.symlinks/plugins/url_launcher_ios/ios`) was resolved to 0.0.1, which depends on
          Flutter
```

But it turned out that everything worked fine after cleaning the repo, deleting the Podfile.lock + Pod folder, updating Flutter (and also fixed the missing bundled Java version for Android Studio - not sure if this had something to do with solving it...) and now it builds fine.

GaelleJoubert, 2022-09-27:
Ok, thank you for your feedback. 
I already tried to clean the repo, to delete  Podfile.lock :/
`Flutter Doctor` tells me my version of Flutter is up to date (Flutter 3.3.2).
I'll try to delete the whole Pod folder and see what it does.

BenelliFurtado, 2022-09-27:
Hmm or maybe you can try, I remember this solved an issue for me in the past
```
pod update iOSDFULibrary
```

GaelleJoubert, 2022-09-27:
I just tried it (with more flutter clean).
No success :/
I also tried to unInstall and re-install Cocoapods, but no more sucess :/

GaelleJoubert, 2022-09-27:
I had a look at what exactly changed in between these versions (for IOS).
And it is just that line in the NordicDFU.Podspec :
` s.dependency 'iOSDFULibrary', '~> 4.11.1'`

which is now :
` s.dependency 'iOSDFULibrary', '~> 4.13.0'`

I had a look at the iOSDFULibrary github changelog :
https://github.com/NordicSemiconductor/IOS-DFU-Library/releases

which said : 
" Breaking: Creating DFUFirmware may now throw an error by @philips77 in https://github.com/NordicSemiconductor/IOS-DFU-Library/pull/486
    Logging improvements by @philips77 in https://github.com/NordicSemiconductor/IOS-DFU-Library/pull/487 "

Which seems to be pretty much related to the error I have.

They offer a migration guide. I'm gonna try it.

In the meantime : I switch the dependency to : ` s.dependency 'iOSDFULibrary', '~> 4.12.0'`

And it works well. So it seems the issue is related to that last version (4.13.0)

GaelleJoubert, 2022-09-27:
If I uderstand well the Issue is exactly because of the breaking change in the DFU Library.
The code is :

```
guard let firmware = DFUFirmware(urlToZipFile: URL(fileURLWithPath: filePath)) else {

                    result(FlutterError(code: "DFU_FIRMWARE_NOT_FOUND", message: "Could not dfu zip file", details: nil))

                    return

                }
```

And the errors say :
```
 "Initializer for conditional binding must haveOptional type, not 'DFUFirmware' 
"Call can throw, but it is not marked with 'try' and the error is not handled"
```
Abd the Breaking change state : "Creating DFUFirmware may now throw an error".
Which is exaclty the error I get. 

I think that somehow the swift must be modified to not use the conditional "guard /else", because now DFU firmware cannot be "nil", and instead use try catch. 

I tried to modify the swift code in that direction, but I have ZERO experiance in swift so I did not suceed yet.

I any one is competant in swift I guess that is a super easy fix.

GaelleJoubert, 2022-09-27:
Ok, I found the fix.
@BenelliFurtado  I think you did not had the last version (5.1.0) install. 
Because the dependency is updated to 4.13.0 and the Breaking change in the changelog maked it IMPOSSIBLE to build without a little change to the `SwiftNordicDFUplugin.swift `:

I will post a pull request, but here is the code I changed (l 93 ) :

```
private func startDfu(
        _ address: String,
        name: String?,
        filePath: String,
        forceDfu: Bool?,
        enableUnsafeExperimentalButtonlessServiceInSecureDfu: Bool?,
        alternativeAdvertisingNameEnabled: Bool?,
        result: @escaping FlutterResult) {                        
                guard let uuid = UUID(uuidString: address) else {
                    result(FlutterError(code: "DEVICE_ADDRESS_ERROR", message: "Device address conver to uuid failed", details: "Device uuid \(address) convert to uuid failed"))
                    return
                }    

//Wrap the following  with a "DO" , remove the "guard"    and add "try " before DFU FIRMWARE
//Put the content of the "else" in a catch later     
            do {

                let firmware = try DFUFirmware(urlToZipFile: URL(fileURLWithPath: filePath))              
                let dfuInitiator = DFUServiceInitiator(queue: nil)
                    .with(firmware: firmware);
                dfuInitiator.delegate = self
                dfuInitiator.progressDelegate = self
                dfuInitiator.logger = self
                
                if let enableUnsafeExperimentalButtonlessServiceInSecureDfu = enableUnsafeExperimentalButtonlessServiceInSecureDfu {

                    dfuInitiator.enableUnsafeExperimentalButtonlessServiceInSecureDfu = enableUnsafeExperimentalButtonlessServiceInSecureDfu
                }              
                if let forceDfu = forceDfu {

                    dfuInitiator.forceDfu = forceDfu
                }               
                if let alternativeAdvertisingNameEnabled = alternativeAdvertisingNameEnabled {

                    dfuInitiator.alternativeAdvertisingNameEnabled = alternativeAdvertisingNameEnabled
                }              
                pendingResult = result
                deviceAddress = address
                dfuController = dfuInitiator.start(targetWithIdentifier: uuid)
            }
        catch{
                result(FlutterError(code: "DFU_FIRMWARE_NOT_FOUND", message: "Could not dfu zip file", details: nil))
                return
            }
      }
```

It fixes the error for me !

GaelleJoubert, 2022-09-27:
@juliansteenbakker the pull request has been posted ! ;)

BenelliFurtado, 2022-09-27:
Ah I see, my Podfile.lock was resolved to use `iOSDFULibrary (4.11.1)` due to my other dependencies. 👍🏼 👍🏼

GaelleJoubert, 2022-09-27:
Aaah, perfect :) Everything makes sense then.

closed 2022-09-28 by juliansteenbakker