juliansteenbakker/nordic_dfu #64: Android upload is excessively slow because uploadPacketDelay is set to 400ms.

corecode, 2022-10-16:
https://github.com/juliansteenbakker/nordic_dfu/blob/master/android/src/main/kotlin/dev/steenbakker/nordicdfu/NordicDfuPlugin.kt#L120 unconditionally sets 400ms intra-packet delay, which severely slows down the firmware upload.  Please remove or allow configuration.

aabdulat1998, 2023-01-31:
Would it be possible to manually go in there and change it?

corecode, 2023-01-31:
That's what I did in my fork.

corecode, 2023-01-31:
Looks like it was fixed: <https://github.com/juliansteenbakker/nordic_dfu/blob/7b97b63f5cc52883dd962e29645cc7e6d6aad098/lib/src/android_special_paramter.dart#L57>.  Just set this value to 0 and it should be much faster.

Thanks @juliansteenbakker!

closed 2023-01-31 by corecode

letisoft, 2023-02-01:
Hello,

I have updated to 5.2.1 but AndroidSpecialParameter still has no delay parameter? am I doing something wrong or fix is not yet in an official release?

Thank you very much!
V