juliansteenbakker/nordic_dfu #75: DFU FAILED: Operation failed, Address: E1D80815-4F69-EA07-D3F0-F49FDEFF95BA, Error type 94

letisoft, 2023-01-05:
Hello,

I started getting following error on iOS:

Error = PlatformException(94, DFU FAILED: Operation failed, Address: E1D80815-4F69-EA07-D3F0-F49FDEFF95BA, Error type 94, null)
#0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:653)
#1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:296)
<asynchronous suspension>
#2      _DeviceUpdateScreenState.doDfu (package:otadfu/ui/exui/updatedevicescreen.dart:142)
<asynchronous suspension>

Thank you very much!

juliansteenbakker, 2023-01-05:
This error has something to do with remoteButtonlessDFU. See https://github.com/NordicSemiconductor/IOS-DFU-Library/blob/c899c507da6fb2527ed78660daacb29093168194/iOSDFULibrary/Classes/Implementation/DFUServiceDelegate.swift#L98

I suggest creating an issue in the above repository, which is what is actually used for iOS DFU.

closed 2023-01-05 by juliansteenbakker

letisoft, 2023-01-16:
Hello,

iosSpecialParameter: const IosSpecialParameter(alternativeAdvertisingNameEnabled: false)

made it work