juliansteenbakker/nordic_dfu #98: Android manifest should not restrict maxSdkVersion

AvivZem, 2023-07-16:
I have an issue when i try to upload my app to google console.
I get the 2 errors: 

1. Duplicate declarations of <uses-permission> element 'android.permission.ACCESS_FINE_LOCATION' in manifest with different maxSdkVersion.
2. Duplicate declarations of <uses-permission> element 'android.permission.ACCESS_COARSE_LOCATION' in manifest with different maxSdkVersion.

it's true that the official docs state that fine location is no longer needed for BLE scanning but i found that i was not able to get BLE scan results unless i have location (fine) enabled.

[I also found that other developers encounter this issue](https://stackoverflow.com/a/70663017/15316543)

My android manifest does not restrict max SDK version, but this library does with the following:

```
<!-- required for API 23 - 30 -->
    <uses-permission-sdk-23
        android:name="android.permission.ACCESS_COARSE_LOCATION"
        android:maxSdkVersion="30" />
    <uses-permission-sdk-23
        android:name="android.permission.ACCESS_FINE_LOCATION"
        android:maxSdkVersion="30" />
``` 

the change that is needed:

```
<uses-permission-sdk-23 android:name="android.permission.ACCESS_COARSE_LOCATION" />
<uses-permission-sdk-23 android:name="android.permission.ACCESS_FINE_LOCATION" />
``` 

For now, my workaround is to edit this library AndroidManifest so i will be able to upload my app to google console.

juliansteenbakker, 2023-07-31:
Merged, will be fixed in the next version.

closed 2023-07-31 by juliansteenbakker

Roee-Tsur, 2023-08-01:
@juliansteenbakker is this fix included in version 6.1.0?
I'm getting the error "Bluetooth permission missing in manifest" from permission handler
I think it's related to this issue.

juliansteenbakker, 2023-08-01:
Yes, i have not changed anything regarding the bluetooth permission in the manifest. Which android SDK are you getting this error? @Roee-Tsur

Roee-Tsur, 2023-08-01:
33