juliansteenbakker/nordic_dfu #100: `PlatformException` -> Code: 133, DFU FAILED: GATT ERROR

shubhamsinghmutualmobile, 2023-07-24:
Hi @juliansteenbakker, I am getting this issue every time I try to start DFU for my BLE device. Could you please help?

```
I/DfuBaseService(12367): DFU service destroyed
E/flutter (12367): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(133, DFU FAILED: GATT ERROR, Address: C3:CE:FE:CF:F5:E9, Error Type: 1, null)
E/flutter (12367): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:652:7)
E/flutter (12367): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:310:18)
E/flutter (12367): <asynchronous suspension>
E/flutter (12367):
```

## Additional info
1. Library version: `6.0.1`
2. Flutter version: `3.13.0-0.2.pre`
3. Android Version: `13 (API 33)`
4. Android Kotlin Version: `1.8.22`

### Complete Logs
```
D/BluetoothAdapter(12367): isLeEnabled(): ON
D/BluetoothLeScanner(12367): onScannerRegistered() - status=0 scannerId=5 mScannerId=0
D/BluetoothAdapter(12367): isLeEnabled(): ON
I/DfuBaseService(12367): DFU service created. Version: 2.3.0
I/DfuBaseService(12367): Starting DFU service in foreground
I/DfuBaseService(12367): Connecting to the device...
D/BluetoothGatt(12367): connect() - device: C3:CE:FE:CF:F5:E9, auto: false
D/BluetoothGatt(12367): registerApp()
D/BluetoothGatt(12367): registerApp() - UUID=832eaa21-d03a-411a-b24d-471893d93208
D/BluetoothGatt(12367): onClientRegistered() - status=0 clientIf=7
D/BluetoothGatt(12367): onClientConnectionState() - status=0 clientIf=7 device=C3:CE:FE:CF:F5:E9
I/DfuBaseService(12367): Connected to GATT server
I/DfuBaseService(12367): Waiting 1600 ms for a possible Service Changed indication...
D/BluetoothGatt(12367): discoverServices() - device: C3:CE:FE:CF:F5:E9
I/DfuBaseService(12367): Attempting to start service discovery... succeed
D/BluetoothGatt(12367): onSearchComplete() = Device=C3:CE:FE:CF:F5:E9 Status=0
I/DfuBaseService(12367): Services discovered
I/DfuImpl (12367): Buttonless service without bond sharing found -> SDK 13 or newer
W/DfuImpl (12367): Device is paired! Use Buttonless DFU with Bond Sharing instead (SDK 14 or newer)
I/DfuImpl (12367): Enabling indications...
D/BluetoothGatt(12367): setCharacteristicNotification() - uuid: 8ec90003-f315-4f60-9fb8-838830daea50 enable: true
I/DfuImpl (12367): Sending Enter Bootloader (Op Code = 1)
I/DfuImpl (12367): Response received (Op Code = 1, Status = 1)
I/DfuBaseService(12367): Disconnecting from the device...
D/BluetoothGatt(12367): cancelOpen() - device: C3:CE:FE:CF:F5:E9
D/BluetoothGatt(12367): onClientConnectionState() - status=0 clientIf=7 device=C3:CE:FE:CF:F5:E9
I/DfuBaseService(12367): Disconnected from GATT server
I/DfuImpl (12367): Device disconnected
W/m.emberbaby.stg(12367): Accessing hidden method Landroid/bluetooth/BluetoothGatt;->refresh()Z (unsupported, reflection, allowed)
D/BluetoothGatt(12367): refresh() - device: C3:CE:FE:CF:F5:E9
I/DfuBaseService(12367): Refreshing result: true
I/DfuBaseService(12367): Cleaning up...
D/BluetoothGatt(12367): cancelOpen() - device: C3:CE:FE:CF:F5:E9
D/BluetoothGatt(12367): close()
D/BluetoothGatt(12367): unregisterApp() - mClientIf=7
I/DfuImpl (12367): Restarting to bootloader mode
D/BluetoothAdapter(12367): isLeEnabled(): ON
D/BluetoothLeScanner(12367): onScannerRegistered() - status=0 scannerId=5 mScannerId=0
D/BluetoothAdapter(12367): isLeEnabled(): ON
I/DfuImpl (12367): Scanning for new address finished with: C3:CE:FE:CF:F5:E9
I/DfuBaseService(12367): Starting DFU service in foreground
I/DfuBaseService(12367): Connecting to the device...
D/BluetoothGatt(12367): connect() - device: C3:CE:FE:CF:F5:E9, auto: false
D/BluetoothGatt(12367): registerApp()
D/BluetoothGatt(12367): registerApp() - UUID=864511eb-3ac0-4b8b-935a-797b07ccb485
D/BluetoothGatt(12367): onClientRegistered() - status=0 clientIf=7
D/BluetoothGatt(12367): onClientConnectionState() - status=0 clientIf=7 device=C3:CE:FE:CF:F5:E9
I/DfuBaseService(12367): Connected to GATT server
I/DfuBaseService(12367): Waiting 1600 ms for a possible Service Changed indication...
D/BluetoothGatt(12367): discoverServices() - device: C3:CE:FE:CF:F5:E9
I/DfuBaseService(12367): Attempting to start service discovery... succeed
D/BluetoothGatt(12367): onConnectionUpdated() - Device=C3:CE:FE:CF:F5:E9 interval=6 latency=0 timeout=500 status=8
D/BluetoothGatt(12367): onConnectionUpdated() - Device=C3:CE:FE:CF:F5:E9 interval=6 latency=0 timeout=500 status=8
D/BluetoothGatt(12367): onClientConnectionState() - status=8 clientIf=6 device=C3:CE:FE:CF:F5:E9
D/FlutterBluePlugin(12367): [onConnectionStateChange] status: 8 newState: 0
D/BluetoothGatt(12367): onClientConnectionState() - status=8 clientIf=7 device=C3:CE:FE:CF:F5:E9
W/DfuBaseService(12367): Target device disconnected with status: 8
I/DfuBaseService(12367): Connection error after: 4087 ms
E/DfuBaseService(12367): An error occurred while connecting to the device: 8
I/DfuBaseService(12367): Attempt: 1
D/BluetoothGatt(12367): refresh() - device: C3:CE:FE:CF:F5:E9
I/DfuBaseService(12367): Refreshing result: true
I/DfuBaseService(12367): Cleaning up...
D/BluetoothGatt(12367): cancelOpen() - device: C3:CE:FE:CF:F5:E9
D/BluetoothGatt(12367): close()
D/BluetoothGatt(12367): unregisterApp() - mClientIf=7
I/DfuBaseService(12367): Restarting the service
I/DfuBaseService(12367): Action received: android.bluetooth.device.action.ACL_DISCONNECTED
I/DfuBaseService(12367): Starting DFU service in foreground
I/DfuBaseService(12367): Connecting to the device...
D/BluetoothGatt(12367): connect() - device: C3:CE:FE:CF:F5:E9, auto: false
D/BluetoothGatt(12367): registerApp()
D/BluetoothGatt(12367): registerApp() - UUID=24d78135-cd9e-4560-b438-004d1494b7f8
D/BluetoothGatt(12367): onClientRegistered() - status=0 clientIf=7
I/TRuntime.CctTransportBackend(12367): Making request to: https://firebaselogging-pa.googleapis.com/v1/firelog/legacy/batchlog
D/TrafficStats(12367): tagSocket(79) with statsTag=0xffffffff, statsUid=-1
D/BluetoothGatt(12367): onClientConnectionState() - status=133 clientIf=7 device=C3:CE:FE:CF:F5:E9
E/DfuBaseService(12367): Connection state change error: 133 newState: 0
I/DfuBaseService(12367): Connection error after: 30025 ms
E/DfuBaseService(12367): Device not reachable. Check if the device with address C3:CE:FE:CF:F5:E9 is in range, is advertising and is connectable
I/DfuBaseService(12367): Attempt: 2
D/BluetoothGatt(12367): refresh() - device: C3:CE:FE:CF:F5:E9
I/DfuBaseService(12367): Refreshing result: true
I/DfuBaseService(12367): Cleaning up...
D/BluetoothGatt(12367): cancelOpen() - device: C3:CE:FE:CF:F5:E9
D/BluetoothGatt(12367): close()
D/BluetoothGatt(12367): unregisterApp() - mClientIf=7
I/DfuBaseService(12367): Restarting the service
I/DfuBaseService(12367): Starting DFU service in foreground
I/DfuBaseService(12367): Connecting to the device...
D/BluetoothGatt(12367): connect() - device: C3:CE:FE:CF:F5:E9, auto: false
D/BluetoothGatt(12367): registerApp()
D/BluetoothGatt(12367): registerApp() - UUID=d2d67250-f09c-426c-bd10-7c280f09bfd3
D/BluetoothGatt(12367): onClientRegistered() - status=0 clientIf=7
I/TRuntime.CctTransportBackend(12367): Status Code: 200
D/BluetoothGatt(12367): onClientConnectionState() - status=133 clientIf=7 device=C3:CE:FE:CF:F5:E9
E/DfuBaseService(12367): Connection state change error: 133 newState: 0
I/DfuBaseService(12367): Connection error after: 30023 ms
E/DfuBaseService(12367): Device not reachable. Check if the device with address C3:CE:FE:CF:F5:E9 is in range, is advertising and is connectable
I/DfuBaseService(12367): Attempt: 3
I/DfuBaseService(12367): Cleaning up...
D/BluetoothGatt(12367): cancelOpen() - device: C3:CE:FE:CF:F5:E9
D/BluetoothGatt(12367): close()
D/BluetoothGatt(12367): unregisterApp() - mClientIf=7
I/DfuBaseService(12367): DFU service destroyed
E/flutter (12367): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(133, DFU FAILED: GATT ERROR, Address: C3:CE:FE:CF:F5:E9, Error Type: 1, null)
E/flutter (12367): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:652:7)
E/flutter (12367): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:310:18)
E/flutter (12367): <asynchronous suspension>
E/flutter (12367): 
D/BluetoothGatt(12367): refresh() - device: C3:CE:FE:CF:F5:E9
D/clearGattCache(12367): CLEAR GATT CACHE: true
D/BluetoothGatt(12367): onClientConnectionState() - status=0 clientIf=6 device=C3:CE:FE:CF:F5:E9
D/FlutterBluePlugin(12367): [onConnectionStateChange] status: 0 newState: 2
D/BluetoothGatt(12367): discoverServices() - device: C3:CE:FE:CF:F5:E9
D/BluetoothGatt(12367): onConnectionUpdated() - Device=C3:CE:FE:CF:F5:E9 interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(12367): onSearchComplete() = Device=C3:CE:FE:CF:F5:E9 Status=0
D/FlutterBluePlugin(12367): [onServicesDiscovered] count: 5 status: 0
```

shubhamsinghshubham777, 2023-07-25:
## Additional info
In the logs, it shows that `Device not reachable. Check if the device with address C3:CE:FE:CF:F5:E9 is in range, is advertising and is connectable` but I can check in the nRFConnect app that while the device is in bootloader state, the MAC address is incremented by 1 i.e. it is `C3:CE:FE:CF:F5:EA` now and the library is not able to search for it on its own, it's still searching for the non-bootloader MAC address.

Any idea about that? What can we do here? Should the library automatically search for a +1 MAC address?

shubhamsinghmutualmobile, 2023-07-25:
## Additional info
In the logs, it shows that `Device not reachable. Check if the device with address C3:CE:FE:CF:F5:E9 is in range, is advertising and is connectable` but I can check in the nRFConnect app that while the device is in bootloader state, the MAC address is incremented by 1 i.e. it is `C3:CE:FE:CF:F5:EA` now and the library is not able to search for it on its own, it's still searching for the non-bootloader MAC address.

Any idea about that? What can we do here? Should the library automatically search for a +1 MAC address?

shubhamsinghmutualmobile, 2023-07-30:
With the help of [philips77](https://github.com/philips77), I was able to find a [solution](https://github.com/NordicSemiconductor/Android-DFU-Library/issues/394#issuecomment-1657129505) for this ticket. We need to use `setRebootTime(long)` to set the device reboot time to more than 0ms (in my case 1000ms works fine) for which I will be raising a new issue/request. Hence, closing this ticket 👍

shubhamsinghmutualmobile, 2023-07-30:
Reopened the issue as it makes more sense to close it after closing https://github.com/juliansteenbakker/nordic_dfu/issues/102

juliansteenbakker, 2023-07-31:
Fixed in v6.1.0

closed 2023-07-31 by juliansteenbakker