juliansteenbakker/nordic_dfu #102: Need `setRebootTime(long)` method

shubhamsinghmutualmobile, 2023-07-30:
In order to resolve [this](https://github.com/juliansteenbakker/nordic_dfu/issues/100) issue, we need to implement the function `setRebootTime(long)` in this library. Using this function, we will be able to delay the device reboot by the provided milliseconds.

Here is the code for the above-said function in the official Android-DFU-Library:

https://github.com/NordicSemiconductor/Android-DFU-Library/blob/b37200033f5d3fe412994e05dd20dd57e14acfe5/lib/dfu/src/main/java/no/nordicsemi/android/dfu/DfuServiceInitiator.java#L299-L302

The expectation is to have a parameter inside both `AndroidSpecialParameter` and `IosSpecialParameter` class as follows:
```dart
firmwareUpdater.startDfu(
    androidSpecialParameter: AndroidSpecialParameter(
        rebootTime: 1000, // Time (int) in millis
    ),
    iosSpecialParameter: IosSpecialParameter(
        rebootTime: 1000, // Time (int) in millis
    ),
);
```

Could you please help with this @juliansteenbakker?

juliansteenbakker, 2023-07-31:
I have implemented this for android, however the ios version does not seem to have this parameter.

juliansteenbakker, 2023-07-31:
Fixed in v6.1.0

closed 2023-07-31 by juliansteenbakker