juliansteenbakker/nordic_dfu #107: Unhandled Exception: PlatformException(6, DFU FAILED: Operation failed, Address: 6D5C4F40-C754-57D4-9D6A-DE57DC88B15F, Error type 6, null)

rohanali1798, 2023-08-04:
Plugin version: 6.1.0
Flutter version:  3.10.4 • channel stable 

I can update the firmware using nRF iOS app but not with my Flutter iOS app. 

In nRF iOS app I had to set the following parameters to successfully do the DFU:

- Allow Resuming DFU: true
- PRN(s): 5
- Data Object Preparation: 0.3s 
- Enable Alternative Advertising Mode: true 
- Force DFU: true
- Enable Experimental Buttonless DFU: false 

In my Flutter app I have the following function that tries to replicate mostly the above parameters:

```
Future<void> startDFU() async
  {
    await NordicDfu().startDfu(
      widget.deviceID,
      'assets/20230727.zip',
      numberOfPackets: 5, //I assume this is same as PNR(s)
      forceDfu: true, 
      onProgressChanged: (
          deviceAddress,
          percent,
          speed,
          avgSpeed,
          currentPart,
          partsTotal,
          ) {
        print('deviceAddress: $deviceAddress, percent: $percent');
      },
    fileInAsset: true,
        onDfuCompleted: handleDfuCompleted,
        onDfuAborted: handleDfuAborted,
        onDfuProcessStarted: (String deviceAddress) =>
            print("DFU PROCESS STARTED"),
        onDeviceConnected: (String deviceAddress) =>
            print("DEVICE CONNECTED"),
        onDeviceConnecting: (String deviceAddress) =>
            print("DEVICE CONNECTING"),
        onDeviceDisconnected: (String deviceAddress) =>
            print("DEVICE DISCONNECTED"),
        onDeviceDisconnecting: handleDeviceDisconnecting,
        onDfuProcessStarting: (String deviceAddress) =>
            print("DFU PROCESS STARTING"),
        onEnablingDfuMode: (String deviceAddress) =>
            print("ENABLING DFU MODE"),
        onFirmwareValidating: (String deviceAddress) =>
            print("FIRMWARE VALIDATING"),
        onError: handleError,);
  }
```

XCode logs:

```
D: [Callback] Central Manager did update state to: Powered ON
V: Connecting to Balumero-LT23XXXX...
D: centralManager.connect(peripheral, options: nil)
flutter: DEVICE CONNECTING
D: [Callback] Central Manager did connect peripheral
I: Connected to Balumero-LT23XXXX
V: Discovering services...
D: peripheral.discoverServices(nil)
I: Services discovered
V: Starting Legacy DFU...
I: Connected to Balumero-LT23XXXX
I: Services discovered
V: Legacy DFU Service found
V: Discovering characteristics in DFU Service...
D: peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
I: DFU characteristics discovered
V: Reading DFU Version number...
D: peripheral.readValue(00001534-1212-EFDE-1523-785FEABCD123)
flutter: DFU PROCESS STARTING
I: Read Response received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x): 0100
A: Version number read: 0.1
V: Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
D: peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
V: Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
A: DFU Control Point notifications enabled
W: Application with buttonless update found
V: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
D: peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
flutter: ENABLING DFU MODE
D: [Callback] Central Manager did disconnect peripheral
I: Disconnected by the remote device
V: Connecting to Balumero-LT23XXXX...
D: centralManager.connect(peripheral, options: nil)
flutter: Disconnected from 6D5C4F40-C754-57D4-9D6A-DE57DC88B15F
D: [Callback] Central Manager did connect peripheral
I: Connected to Balumero-LT23XXXX
V: Discovering services...
D: peripheral.discoverServices([00001530-1212-EFDE-1523-785FEABCD123])
I: Services discovered
V: Legacy DFU Service found
V: Discovering characteristics in DFU Service...
D: peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
I: DFU characteristics discovered
V: Reading DFU Version number...
D: peripheral.readValue(00001534-1212-EFDE-1523-785FEABCD123)
I: Read Response received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x): 0800
A: Version number read: 0.8
V: Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
D: peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
flutter: DFU PROCESS STARTING
V: Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
A: DFU Control Point notifications enabled
V: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
D: peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
V: Writing image sizes (0b, 0b, 161024b) to characteristic 00001532-1212-EFDE-1523-785FEABCD123...
D: peripheral.writeValue(0x000000000000000000750200, for: 00001532-1212-EFDE-1523-785FEABCD123, type: .withoutResponse)
I: Data written to 00001531-1212-EFDE-1523-785FEABCD123
A: Start DFU (Op Code = 1, Upload Mode = 4) request sent
I: Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100101
A: Response (Op Code = Start DFU, Status = Success) received
A: Writing Initialize DFU Parameters (Op Code = 2, Type = Begin)...
V: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
D: peripheral.writeValue(0x0200, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
V: Writing to characteristic 00001532-1212-EFDE-1523-785FEABCD123...
D: peripheral.writeValue(0x5200ffffffffffff0100b6000f71, for: 00001532-1212-EFDE-1523-785FEABCD123, type: .withoutResponse)
V: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
D: peripheral.writeValue(0x0201, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
I: Data written to 00001531-1212-EFDE-1523-785FEABCD123
I: Data written to 00001531-1212-EFDE-1523-785FEABCD123
I: Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100201
A: Response (Op Code = Initialize DFU Parameters, Status = Success) received
A: Initialize DFU Parameters completed
V: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
D: peripheral.writeValue(0x080c00, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
I: Data written to 00001531-1212-EFDE-1523-785FEABCD123
A: Packet Receipt Notif Req (Op Code = 8, Value = 12) request sent
V: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
D: peripheral.writeValue(0x03, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
I: Data written to 00001531-1212-EFDE-1523-785FEABCD123
A: Uploading firmware...
V: Sending firmware to DFU Packet characteristic...
flutter: deviceAddress: 6D5C4F40-C754-57D4-9D6A-DE57DC88B15F, percent: 0
I: Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100306
A: Response (Op Code = Receive Firmware Image, Status = Operation failed) received
E: Error 6: Operation failed
V: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
D: peripheral.writeValue(0x06, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
D: [Callback] Central Manager did disconnect peripheral
I: Disconnected by the remote device
flutter: ERROR: Operation failed
[VERBOSE-2:dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(6, DFU FAILED: Operation failed, Address: 6D5C4F40-C754-57D4-9D6A-DE57DC88B15F, Error type 6, null)
#0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:652:7)
#1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:310:18)
<asynchronous suspension>
#2      _SettingsState.startDFU (package:my_flutter_reactive_ble/settings.dart:106:5)
<asynchronous suspension>
#3      _SettingsState.build.<anonymous closure> (package:my_flutter_reactive_ble/settings.dart:82:5)
<asynchronous suspension>
```

From the logs, everything looks fine until when DFU actually starts to upload and then it throws exception at 0%.

samnbd, 2023-12-14:
Having the same issue on iOS devices

rohanali1798, 2024-04-03:
> Having the same issue on iOS devices
> 
> 
Hi @samnbd are you able to do it on Android? I haven't tested it on Android though.

rohanali1798, 2024-04-03:
@juliansteenbakker 
Any plans to work on this issue?

juliansteenbakker, 2026-08-17:
Please let me know if this issue still exists in the latest v7.2.0

closed 2026-08-17 by juliansteenbakker