juliansteenbakker/nordic_dfu #110: Nothing happens when calling startDfu

Roee-Tsur, 2023-08-07:
Hey,
I'm trying to implement the plugin in the most basic way but I get no logs and my callbacks are not called.
What might be the problem?

This is my implementation: 

```
if (selectedDfuPath == null) {
                          Fluttertoast.showToast(msg: 'no file selected');
                          return;
                        }
                        String? address = device.getMacAddress();
                        if (address == null) {
                          Fluttertoast.showToast(msg: 'mac address is null');
                          return;
                        }
                        NordicDfu().startDfu(
                          address,
                          selectedDfuPath!,
                          onProgressChanged: (String address,
                              int percent,
                              double speed,
                              double avgSpeed,
                              int currentPart,
                              int totalParts) {
                            setState(() {
                              dfuProgress = percent;
                            });
                          },
                          onDfuCompleted: (String address) =>
                              Fluttertoast.showToast(
                                  msg: 'completed dfu: $address'),
                          onError: (String address, int error, int errorType,
                                  String message) =>
                              Fluttertoast.showToast(
                                  toastLength: Toast.LENGTH_LONG,
                                  msg:
                                      'ERROR:\naddress: $address\nerror: $error\nerror type: $errorType\nmessage: $message'),
                        );
```

rstewart22, 2023-08-07:
I'm experiencing this issue also on Android, but iOS works

JohanneBoisvilliers, 2023-08-18:
even in 6.0.1 I have the same issue, no callbacks called

rstewart22, 2023-08-18:
I found this not working at 6.0.1 on a certain device (Galaxy Tab A) but worked on others

juliansteenbakker, 2023-08-29:
This issue has been fixed in v6.1.2

closed 2023-08-29 by juliansteenbakker