juliansteenbakker/nordic_dfu #115: Error on build app

andreamainella98, 2023-09-04:
Hello, when I try to run the production build, I get the following error:

```shell
FAILURE: Build failed with an exception.

What went wrong:
Execution failed for task ':nordic_dfu:compileReleaseJavaWithJavac'.
error: invalid source release: 17
```

The version of Flutter we are using is the following, using FVM:
```shell
[✓] Flutter (Channel stable, 3.0.5, on macOS 13.4.1 22F770820d darwin-arm, locale it-IT)
```

How can I resolve this issue? Because when I remove your plugin, it works correctly 😅

HowManyOliversAreThere, 2023-09-15:
Encountered the same issue on Android when running 6.1.2 (latest), works fine on iOS.

Android known to be previously working on 6.0.1, reverting to this version was our solution - obviously not a very good one...

juliansteenbakker, 2023-10-09:
Please use java 17 or higher for this project: https://stackoverflow.com/a/70215091/13031778

closed 2023-10-09 by juliansteenbakker