juliansteenbakker/nordic_dfu #141: Constant DFU_FIRMWARE_NOT_FOUND error

amatier, 2024-03-24:
I'm running the most up-to-date versions of flutter as well as versions of this package.

I'm making my call like this:
```
 onPressed: () async {
                              final notifier =
                                  ref.read(firmwareNotifierProvider.notifier);
                              final path = await notifier.getFirmwarePath(
                                  latestFirmware.downloadUrl ?? '');
                              print('Firmware path: $path');
                              print('Device remoteId: ${uiState.remoteId}');
                              await NordicDfu().startDfu(
                                uiState.remoteId,
                                path,
                              );
                            },
```
Where path is set by downloading a .zip file from Firebase and placing it in a temp directory:
```
Future<String> getFirmwarePath(String url) async {
    // Get the temporary directory
    Directory tempDir = await getTemporaryDirectory();
    // Define the local file path
    String filePath = '${tempDir.path}/file.zip';

    // Use Dio to download the file
    Dio dio = Dio();
    await dio.download(url, filePath);

    if (await File(filePath).exists()) {
      print('Download successful, file is where it should be.');
      return filePath;
    } else {
      throw Exception("Download failed, file's not in the temp directory.");
    }
  }
 ```
 
 I received a path back and it appears that half works correctly. The problem is when calling the `.startDfu()` it fails 10/10 times. I receive this error:
 ```
 flutter: ----------------FIREBASE CRASHLYTICS----------------
flutter: PlatformException(DFU_FIRMWARE_NOT_FOUND, Could not dfu zip file, null, null)
flutter: 
#0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:651:7)
#1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:334:18)
<asynchronous suspension>
#2      KeypadDetailsBluetooth.build.<anonymous closure>.<anonymous closure> (package:*project name removed*/features/admin/displays/keypad_settings.dart:319:31)
<asynchronous suspension>
flutter: ----------------------------------------------------
```

The content of my .zip file is a .bin file named `app_update.bin` and a manifest file called `manifest.json` I've tried just passing the .bin file and that fails too. I've tried putting my zip file into assets and going that route and it still fails. I've tried using my .zip file with NRF connect and it works flawlessly. I'm not sure why this is happening here because the error message isn't explicit enough in what it is trying to portray.

Any help is much appreciated.

joaquin03, 2024-07-20:
@amatier, did you manage to solve this issue?

amatier, 2024-07-22:
> @amatier, did you manage to solve this issue?

I did not, I ended up using a different package to achieve what I wanted to do. 
I ended up using [mcumgr_flutter](https://pub.dev/packages/mcumgr_flutter)

danww, 2025-02-25:
I had this issue; the problem was that the file - as stored locally on the mobile - did not have a '.zip' file extension.
Appending a '.zip' extension to the firmware file allowed the DFU process to work and complete successfully.

Note that it always worked fine on Android devices, with or without the '.zip' extension.

xShuusui, 2025-05-19:
@danww I would be very carefully with the .zip extension. I can agree that it always worked when the dfu zip only contained the firmware itself. But when having multiple bins within manifest the Android DFU Library can not handle it correctly. 

See https://github.com/juliansteenbakker/nordic_dfu/issues/206 where this package copies and removes the file extension. We are not able to update softdevice+bootloader+firmware together without .zip extension.

closed 2026-08-18 by juliansteenbakker