juliansteenbakker/nordic_dfu #159: DFU Fails on iOS but Succeeds on Android (DFU Error Code : 302)

Soulddgy, 2024-08-08:
I'm experiencing an issue with DFU on iOS. While the DFU process works correctly on Android, it fails on iOS. I am using the same MAC address and UUID for both the bootloader and Bluetooth device, which seems to be causing the problem.
On iOS, I receive a "DFU Service not found" error. I suspect this is related to the way iOS handles Bluetooth connections. Any guidance or suggestions would be greatly appreciated.

console : 

flutter: remote id for dfu : 6949E5E4-3A02-F8AA-AD92-B9DE3583AE72
flutter: 1-2024-08-08 13:50:56.772355 -- PublishingManager::publish - entered with topic D1
flutter: DFU error: PlatformException(302, DFU FAILED: DFU Service not found, Address: 6949E5E4-3A02-F8AA-AD92-B9DE3583AE72, Error type 302, null)

And this is my startDfu code :

[image]

shubhamsinghshubham777, 2024-11-07:
Try disabling `alternativeAdvertisingNameEnabled` in `iosSpecialParameter` like so:
```dart
final result = await NordicDfu().startDfu(
  remoteId,
  filePath,
  fileInAsset: false,,
  forceDfu: true,
  iosSpecialParameter: IosSpecialParameter(
    alternativeAdvertisingNameEnabled: false,
  ),
  enableUnsafeExperimentalButtonlessServiceInSecureDfu: true,
  progressChanged: (
    _,
    percent,
    __,
    ___,
    ____,
    _____
  ) {
    progress.value = percent / 100.0;
  },
);
print('DFU result: $result');
```

juliansteenbakker, 2026-08-17:
Please let me know if this issue still exists in the latest v7.2.0

closed 2026-08-17 by juliansteenbakker