juliansteenbakker/nordic_dfu #163: Remove all permissions from AndroidManifest.xml to let users choose what they want

shubhamsinghshubham777, 2024-08-15:
In the current implementation, the library's `AndroidManifest.xml` contains all permissions required by the library to function properly. That is good, but now the users do not have the flexibility to remove any permissions they do not need. For example, I do not want `FINE_LOCATION` and `COARSE_LOCATION` permissions on Android 12 and above since they are not a hard requirement anymore because I can assert that by using `android:usesPermissionFlags="neverForLocation"`, but the library still forces me to have these permissions in my merged manifest. So, if I try to override the permissions by using `tools:maxSdkVersion=30`, it not only does not work but also causes the Play Store console to throw the following issues when I try to upload an `APK` or `AAB`:

```
Duplicate declarations of permission android.permission.ACCESS_FINE_LOCATION with different maxSdkVersions
Duplicate declarations of permission android.permission.ACCESS_COARSE_LOCATION with different maxSdkVersions
```

[image]

Therefore, I request that we please remove all permissions from the library's AndroidManifest.xml and update the README to instead guide the users to add in only the permissions they require. I am happy to lend a hand and raise a PR if it is acceptable, do let me know.

Just in case someone is trying to find a temporary solution/workaround for this Google Play Store issue, you can follow [this](https://stackoverflow.com/a/32144838/20325172) solution:

[image]

In this case, the `selector` would be `dev.steenbakker.nordicdfu`

danww, 2024-09-23:
I'm struggling with this as well.
We've got explicit settings for Fine and Coarse Location permissions, as follows:

```
<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" android:maxSdkVersion="30"/>
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" android:maxSdkVersion="28" />
```

...but I can't seem to get the workaround, mentioned above, to work.

I then get the following error:

```
Publishing failed :|
Google Play failed to upload artefacts. You must let us know whether your app uses any Foreground Service permissions.: {
    "error": {
        "code": 403,
        "message": "You must let us know whether your app uses any Foreground Service permissions.",
        "status": "PERMISSION_DENIED"
    }
}
```

shubhamsinghshubham777, 2024-09-28:
> I'm struggling with this as well. We've got explicit settings for Fine and Coarse Location permissions, as follows:
>
> ```
> <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" [...]

I'm not entirely sure about this, but seems like this error is occurring because of some library you might have added that declares a `FOREGROUND_SERVICE` permission and Google wants you to confirm whether you actually need this foreground service permission (or you have mistakenly added such a transitive permission) as they seem to have been getting quite sensitive about foreground services over the past few years.

You can see an answer here: https://stackoverflow.com/a/77642329

So please check your `merged manifest` and see if some library adds the `FOREGROUND_SERVICE` permission, and if you need it, then explicitly add that permission to your main manifest, otherwise use `tools:node="remove"` in the `FOREGROUND_SERVICE` permission you explicitly add in your main manifest.

Hope this helps somehow

Robbinb1993, 2024-10-10:
My bluetooth permissions in the AndroidManifest file are as follows:

    
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN" android:usesPermissionFlags="neverForLocation"/>
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" tools:node="remove"/>
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
    
    
    <uses-permission android:name="android.permission.BLUETOOTH" android:maxSdkVersion="30" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" android:maxSdkVersion="30" />
    
    
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" tools:node="remove"/>
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" android:maxSdkVersion="28" />
    
    so I am using tools:node to remove any existing ones before declaring my own with the specific max SDK version.

closed 2026-08-18 by juliansteenbakker