juliansteenbakker/nordic_dfu #178: [Android] Handle errors better

shubhamsinghmutualmobile, 2024-11-28:
I see the following crashes in my app's crashlytics window (on Android only):

1. [image]
2. [image]

But since I have wrapped my `startDfu()` method call inside a `try-catch` block, I expected this error to be handled and not crash the app. This would've been true if the `startDfu` future caught the error in the first place, but it seemingly does not.

Ideally, this error should be passed by the library to its own `onError` callback.

shubhamsinghmutualmobile, 2025-02-06:
@juliansteenbakker Could you please help with this?

juliansteenbakker, 2026-08-17:
Start is now wrapped into DFU_START_ERROR (:137) and onError forwards to pendingResults[...].error (:221), so the future does receive it in v7.2.0

closed 2026-08-17 by juliansteenbakker