juliansteenbakker/nordic_dfu #206: [Android] Missing zip extension when copying from asset path to file system

xShuusui, 2025-05-19:
The [Android-DFU-Library](https://github.com/NordicSemiconductor/Android-DFU-Library) requires dfu zip files to have the .zip extension. For Android the dfu zip will be copied to the file system with a temporary timestamp name. This new file name misses the .zip extension and therefore results in a wrong update process. 

The update of only a **firmware** worked magically, but when the dfu file contains **softdevice+bootloader+firmware** we could not update these out of the asset path. It never started the updated process for softdevice and bootloader!

Version: `v7.0.0`

xShuusui, 2025-05-23:
Found the problematic part inside the library [DfuBaseService Line-1157](https://github.com/NordicSemiconductor/Android-DFU-Library/blob/main/lib/dfu/src/main/java/no/nordicsemi/android/dfu/DfuBaseService.java#L1157). A missing .zip ending sets `TYPE_APPLICATION` and not `TYPE_AUTO`, therefore it never tries to update anything else.

FYI @juliansteenbakker

oi-narendra, 2025-09-07:
Same issue. Update never starts with `softdevice+bootloader+firmware` DFU.

closed 2025-09-19 by juliansteenbakker