juliansteenbakker/nordic_dfu #222: [Android] DfuEventHandler callback issue

dslee48, 2025-09-25:
Some of the handlers in the DfuEventHandler class are not working. 

For example, important handlers such as onDfuCompleted and onProgressChanged are not actually triggered—the callback functions are not called—while the firmware update messages through DFU can still be observed internally.

It has been tested on the latest Android OS 15 and iOS 18 devices.

When version 7.0.0 was first released, I also tested it and thought the issue would be resolved soon, so I did not report it. However, it is a bit disappointing that the issue still remains unaddressed.

melihkarakelle, 2025-11-16:
Same here, simply failing on Android.

juliansteenbakker, 2025-11-17:
Sorry for the delay, will take a look at it later today.

juliansteenbakker, 2025-11-17:
@guominfang is this with the code from #227 ?

juliansteenbakker, 2025-11-18:
@guominfang i now understand the need for adress mapping.

Can you check out the new code in the branch? I've updated it to support adress mapping: #227